isg-stuttgart/vscode-isg-cnc #6: Bug with "format code" function on blocs number with colon

AurelWM, 2021-11-12:
**Bug observed**
Hallo, 
Thank you for this extension, I just discovered it and I clean my *.nc programs for a Beckhoff CNC with your function "format code".
I observed the following bug:
If in a program there is a label followed by a colon like:
N10:
The function "format code" add space between label and colon like:
N10   : 
And then if you execute "format code" once again, it add again space.

**Screenshots**
Initial state:
[image]

First execution of "format code":
[image]

Second execution of "format code":
[image]

Thank you, best regards

AurelWM, 2021-11-12:
A important information: it occurs only if the bloc number is inside an indented area.
Between $IF and $ENDIF for example.

andreashafner-isg, 2021-11-12:
Hello @AurelWM,
thank you for reporting the bug and for using the extension. 
I confirmed the bug and I will fix it as soon as possible.

Thank you and best regards.
Andreas

andreashafner-isg, 2021-11-12:
Fixed in [release version v0.2.4](https://github.com/isg-stuttgart/vscode-isg-cnc/releases/tag/V0.2.4)

PR #7 (2021-11-12): 6 Bug with "format code" function on blocs number with colon
# Pull Request

## Description

Fixed format code, add blocknumbers, remove blocknumbers functions in case of nc labels like N10:. 
Formating code calculates now correct indent.
Remove blocknumbers did not delete nc label blocknumbers anymore.
NC Labels will not be changed by adding blocknumbers function.

Fixes #6

## Type of change

- [ x] Bug fix (non-breaking change which fixes an issue)

## How Has This Been Tested?

Tested Format code, add blocknumbers, remove blocknumbers with following code:

```isg-nc
%goto
N05 P1=1
N06 P2=1 [... cut]
closed 2021-11-12 by andreashafner-isg