isg-stuttgart/vscode-isg-cnc #18: Syntax highlight without ISG Themes

mpacher, 2022-02-11:
I would like to ask to make the syntax highlight work without forcing the user to select the "ISG Light/Dark" themes. To me this is really disturbing because I have different projects written in different programming languages and I do not want to be forced in choosing your themes.
Many thanks in advance.

georgulbrich-isg, 2022-02-14:
Hello mpacher,

we will look into alternative ways of syntax highlighting, without forcing the user to switch themes.

Thank you for your suggestion and best regards,
Georg Ulbrich

lukashettler-isg, 2022-03-02:
@mpacher 

In https://github.com/isg-stuttgart/vscode-isg-cnc/pull/22 we discuss about the default syntax highlighting groups. Would you agree with these groups or do you have other ideas or wishes how the groups should be defined?

PR #22 (2022-03-08): changed scopes to standard to use themes
The scopes in the tmlanguage file for cnc have been changed to standard scopes with additional subranges. Now you can use other themes because the range falls back to the known range. With ISG themes it is more granular than with standard themes. for example GCode and MCode in the ISG themes are distinguished again in the coloring. This does not happen in themes like Dark or Dark+.

This will close #18
closed 2022-03-08 by lukashettler-isg