isg-stuttgart/vscode-isg-cnc #29: Function for aligning the equal sign in assignments

andreas-hafner, 2022-03-16:
I tried using the format function, after defining some variables to align the values to the equal signs. Unfortunately, this did not work. Would it be possible that a function is created to format a selected text at the equal sign?
Here is an example:

Before:
```
V.L.MYVAR = 0
V.L.MYSECONDVAR = 100
```

Desired result:
```
V.L.MYVAR       = 0
V.L.MYSECONDVAR = 100
```

lukashettler-isg, 2022-03-18:
Thank you @andreas-hafner for reporting this enhancement. We will consider this request in our further developments.

jurekseverin-isg, 2023-01-04:
This was added as an extra command now, so that this is not done by "Beautify" if you want to align your equal signs otherwise
See: https://github.com/isg-stuttgart/vscode-isg-cnc/pull/98

closed 2023-01-04 by jurekseverin-isg