isg-stuttgart/vscode-isg-cnc #30: isg-commands not found before open nc-file

jurekseverin-isg, 2022-04-28:
**Describe the bug**
If you open a new folder for the first time (or you have no folder opened), while the extension is enabled you cannot call the extension ISG-commands until you open a .nc-file (maybe also some other filetypes but .txt for example did NOT work). If you try call an extension command before doing so an error is thrown: Example: "Command 'ISG-CNC: Find next TFS code' resulted in an error (command 'isg-cnc.FindNextTFS' not found)".
Somehow if you open an .nc file, then close it and then try to use an isg-command there is no error, just no reaction.

**To Reproduce**
Steps to reproduce the behavior:
1. Make sure to have the extension enabled
2. Optional: Open a folder
3. Press ctrl+shift+P and call any isg-command
4.See Error: "Command<commandnameForUser> resulted in an error (command <commandnameIntern> not found)". <commandnameForUser> and <commandnameIntern> are placeholders for the specific commands.

5. Open .nc file
6. Press ctrl+shift+P and call any isg-command
7. See expected command behaviour

8. close .nc file
9. Press ctrl+shift+P and call any isg-command
10. See no reaction

**Expected behavior**
Of course most of these commands aren't useful to call without a nc file being open. However there should not be a "not found error" but just no reaction or maybe a message like " <commandname> can only be used with an opened .nc file".

jurekseverin-isg, 2022-06-30:
- I fixed this while implementing in #16

closed 2022-06-30 by jurekseverin-isg

lukashettler-isg, 2022-07-19:
Hi Jurek, the description of this bugfix is missing in the Changelog.md-File. Can you please add this in the develop branch. Thank you

jurekseverin-isg, 2022-07-19:
> Hi Jurek, the description of this bugfix is missing in the Changelog.md-File. Can you please add this in the develop branch. Thank you

Hi Lukas, i added this in the develop branch.