isg-stuttgart/vscode-isg-cnc #31: ISG-Kernel Documentation-Website not found because of new website

jurekseverin-isg, 2022-06-28:
**Describe the bug**
If you call the startDocu()-Command of the Extension you will land on https://www.isg-stuttgart.de/kernel-html5/de-DE/index.html which says: Page not found.

**Expected behavior**
Instead the new ISG-Kernel Documentation-website should be called: https://www.isg-stuttgart.de/fileadmin/kernel/kernel-html/de-DE/index.html

lukashettler-isg, 2022-06-29:
Hi Jurek,
this problem seems to be a general problem of our website. The person I have to ask is not at work today so I will give you a more detailed answer tomorrow.

If the first link with the help for the marked command is not working, we have to use the defaul link, like you suggested. But it would be better if the link would work...

lukashettler-isg, 2022-06-30:
There is a problem in the website. So we will use the default page until this problem is fixed.

closed 2022-07-05 by lukashettler-isg