isg-stuttgart/vscode-isg-cnc #32: Why Configuration Setting for docu-path?

jurekseverin-isg, 2022-06-28:
**Describe the bug**
The Extension Configuration Settings contain an "Documentation" setting where you can define the path to the ISG documentation.
This is never used in the extension and also not useful because why would you re-define it?

**Suggestion**
Just remove this Configuration Setting

jurekseverin-isg, 2022-06-29:
Same goes with "Browser-linux" and "Browser-windows" configurations. They are never used. Furthermore the npm open package recognizes the users default browser. So there shouldn't be a use-case for these configurations in the future too.

lukashettler-isg, 2022-06-30:
Hi Jurek,

the reason for the setting for the path to the ISG Documentation is that you can reference to a local Documentation if you don't have internet connection. This setting is useful and shouldn't be deleted. But I can't remember the reason for the Browser setting so this can be deleted in my opinion.

jurekseverin-isg, 2022-06-30:
Thanks for feedback. I have only removed the Browser-paths now

closed 2022-06-30 by jurekseverin-isg

lukashettler-isg, 2022-07-05:
@jurekseverin-isg 
Can you please verify that the changes are merged into the develop branch.

jurekseverin-isg, 2022-07-05:
> @jurekseverin-isg Can you please verify that the changes are merged into the develop branch.

The changes were merged succesfully into the develop branch by this Pull-request:
https://github.com/isg-stuttgart/vscode-isg-cnc/pull/36