isg-stuttgart/vscode-isg-cnc #40: language server for ISG G-code

lukashettler-isg, 2022-07-05:


lukashettler-isg, 2022-07-05:
### example grammatic in PEG.js

Link to PEG.js https://pegjs.org/

// Simple Arithmetics Grammar
// ==========================
//
// Accepts expressions like "2 * (3 + 4)" and computes their value.

start
  = program

program
  = title blocks
  
blocks
  = block*
  
block
  = "N" [^T]* tool? .*

title
  = "%" name:[a-zA-Z]+ [\n] 

tool
  = "T" tool_nr:integer { console.log(tool_nr); }

integer "integer"
  = digits:[0-9]+ { return parseInt(digits.join(""), 10); }

_ "whitespace"
  = [ \t\n\r]*

### Example NC-programm for tool change

%Testprogram
N01 G01 X10 Y2 F100
N02 T2
N03 G0 X2
N04 T4
N05 M30

lukashettler-isg, 2022-07-05:
see also https://github.com/isg-stuttgart/vscode-isg-cnc/issues/17

jurekseverin-isg, 2022-07-05:
I will address this issue in a seperate repo: https://github.com/isg-stuttgart/cnc-language-server

closed 2022-08-30 by jurekseverin-isg