isg-stuttgart/vscode-isg-cnc #52: Large file handling for sidebar view

isg-stuttgart-github, 2022-08-23:
**Describe the bug**

If a nc-file contain more than for example 15000 matching lines the extension stops to avoid memory overflow.

jurekseverin-isg, 2022-09-28:
Improved performance a bit, more should'nt be necessary because normally we dont have that much matchings.
Few hundred matches only took like one two seconds. 7000 took like 30sec, should be fine

closed 2022-09-28 by jurekseverin-isg

lukashettler-isg, 2022-09-29:
> Improved performance a bit, more should'nt be necessary because normally we dont have that much matchings. Few hundred matches only took like one two seconds. 7000 took like 30sec, should be fine

Hi @jurekseverin-isg 
the problem was not only the performance. You described that the extension stops to avoid memory overflow if there are to much matches. How is the behaviour for this now?

jurekseverin-isg, 2022-09-29:
> Hi @jurekseverin-isg the problem was not only the performance. You described that the extension stops to avoid memory overflow if there are to much matches. How is the behaviour for this now?

I couldn't reproduce this but i remember what you mean. Only thing what still happens is that other extension utility (seems to include not only our but also other extensions) are blocked while creating the sidebar.
The only possibility I see here to protect the user is cancelling after a specfied time, but I'm not sure if that's what we want.