isg-stuttgart/vscode-isg-cnc #54: Building the extension won't contain required .js files because they don't land in the outDir

jurekseverin-isg, 2022-08-23:


jurekseverin-isg, 2022-08-23:
Hi @lukashettler-isg ,
the small [change](https://github.com/isg-stuttgart/vscode-isg-cnc/compare/main...54-building-the-extension-wont-contain-required-js-files-because-they-dont-land-in-the-outdir) i've done in the belonging branch should fix our issues.
The ts-compiler only includes .ts files in the build per default, but now in the tsconfig.json i specifed to also include .js files.
I've tested building the extension when having empty out-dir and it worked (also included ncParser.js in out-dir).
Hopfully this has finally done it.

closed 2022-08-24 by isg-stuttgart-github

isg-stuttgart-github, 2022-08-24:
Hi @jurekseverin-isg ,
I also thought it would fix the version, but unfortunatly the problem is still there...
We will look for the problem tomorrow.