isg-stuttgart/vscode-isg-cnc #56: isg-commands didnt work on linux system

andreas-hafner, 2022-08-24:
**Describe the bug**
There is no effect if I run any command in context menu or in command line.

**To Reproduce**
Right click for context menu
Click on isg-commands
Click any command you like

**Expected behavior**
running the command

**Desktop (please complete the following information):**
 - OS: Arch Linux
 - 
[image]

lukashettler-isg, 2022-08-25:
Thank you @andreas-hafner for reporting this issue. The behaviour is the same in Windows. We search the solution for this problem.

lukashettler-isg, 2022-08-25:
hi @andreas-hafner we got the extension working again in the new version V0.3.4 but we had to temporarily remove the "start ISG documentation" feature, because this causes the problem. We will fix this in the next version V0.3.5.

closed 2022-08-25 by lukashettler-isg