isg-stuttgart/vscode-isg-cnc #76: Block numbers in combination with skip lines

lukashettler-isg, 2022-10-27:
Die Integration von automatischen Satznummern funktioniert gut und wird auch permanent benutzt.
Bei Verwenden von Skipping müssen die Zeilen, die ausgeblendet werden sollen, im Nachgang manuell bearbeitet werden.
Vorher:

```
%skip_test.nc
G0 X0 Y0 Z0
G1 F100
G1 X2
#TIME 2
#FLUSH WAIT
/Z3
#TIME 2
G1 X4
#FLUSH WAIT
/Y2
X6

M30
```
Nachher:

```
%skip_test.nc
N010 G0 X0 Y0 Z0
N020 G1 F100
N030 G1 X2
N040 #TIME 2
N050 #FLUSH WAIT
N060 /Z3
N070 #TIME 2
N080 G1 X4
N090 #FLUSH WAIT
N100 /Y2
N110 X6

N120 M30
```

Anwenderfreundlich wäre das Ergebnis:
**/N060 Z3**

jurekseverin-isg, 2022-11-23:
@lukashettler-isg sollten Zuweisungszeilen in #VAR-#ENDVAR Böcken nummeriert werden?
Also soll 
```
#VAR
V.L.SurfacePosition = 0
#ENDVAR
```
zu
```
N010 #VAR
N020 V.L.SurfacePosition = 0
N030 #ENDVAR
```
oder zu 
```
N010 #VAR
V.L.SurfacePosition = 0
N020 #ENDVAR
```
werden? Oder nochmal anders?

jurekseverin-isg, 2022-12-02:
I will also replace current Insertion Position by using the parser instead of regexs and indexOf. This allows more flexibility e.g. if you already have a blocknumber behing "/" which should be replaced.

jurekseverin-isg, 2022-12-23:
Closed by https://github.com/isg-stuttgart/vscode-isg-cnc/pull/94

closed 2022-12-23 by jurekseverin-isg