isg-stuttgart/vscode-isg-cnc #91: Add ISG Cycles to the extension

lukashettler-isg, 2022-12-02:
## Desired behaviour

It would be a good feature if the extension would have more informations about the ISG Cycles. If the Cycles could be programmed with snippets to find faster the correct cycle name and it's parameters. Additionally a description of the cycle would be useful or a proposed link to the official documentation.

lukashettler-isg, 2024-02-20:
## Ideas for development

- Snippets including Parameters with language server. Informations are in json files of cnc-cycles. Hover informations.
- Hoverinformation with link to HTML long cycle description

What do we need for this:
- one json file with all cnc-cycles with name and additional the node id of ST4

jurekseverin-isg, 2024-05-06:
Completed with https://github.com/isg-stuttgart/vscode-isg-cnc/pull/152

closed 2024-05-06 by jurekseverin-isg