isg-stuttgart/vscode-isg-cnc #108: "Go to reference" for program declarations, goto-labels and local variable declarations

jurekseverin-isg, 2023-03-20:
Implement the "Go To Reference" feature for the mentioned syntax elements.

jurekseverin-isg, 2023-03-25:
@georgulbrich-isg @lukashettler-isg 
Do we want to include the definition line itself in the listing of references?
E.g. should we insert `%L cycle` at the position of the drawed arrow?
[image]
Pro: User can use this to jump to the definition without more clicks after scrolling through references) and at least the default Typescript Language Server seems to include them (see following function-definition)
[image]
Contra: In my opinion "Definitions" are no "References", as others also think: https://github.com/microsoft/vscode/issues/68071

lukashettler-isg, 2023-03-27:
> @georgulbrich-isg @lukashettler-isg Do we want to include the definition line itself in the listing of references? E.g. should we insert `%L cycle` at the position of the drawed arrow? [image] Pro: User can use this to jump to the definition without more clicks after scrolling through references) and at least the default Typescript Language Server seems to include them (see following function-definition) [image] Contra: In my opinion "Definitions" are no "References", as others also think: [microsoft/vscode#68071](https://github.com/microsoft/vscode/issues/68071)

I would prefer the list without the definition because we have the _Go to definition_ feature. This should be comfortable enough to find the definition fast. And the definition is not a reference...

georgulbrich-isg, 2023-03-27:
Either way fine with me.

jurekseverin-isg, 2023-03-27:
Close as completed in https://github.com/isg-stuttgart/vscode-isg-cnc/pull/115

closed 2023-03-27 by jurekseverin-isg