isg-stuttgart/vscode-isg-cnc #149: Go to fileoffset in selected file

lukashettler-isg, 2024-02-22:
This feature should find a cnc program in the current working space and jump directly to the given fileoffset. With multi selection enabled in VS Code it's possible to mark the cnc program and the fileoffset. The feature should take this information and jump to the program. In the following picture an example is shown, how the input for the functionality could be selected.

[image]

lukashettler-isg, 2024-02-22:
@rosariotaormina-isg 
 fit the description above to your requirements?

jurekseverin-isg, 2024-05-06:
Feature completed with https://github.com/isg-stuttgart/vscode-isg-cnc/pull/156

closed 2024-05-06 by jurekseverin-isg