isg-stuttgart/vscode-isg-cnc #165: File missing docs\images\subProgramComment.png

asteding, 2024-11-23:
**Describe the bug**
A file `docs\images\subProgramComment.png` is missing causing a 404.

**To Reproduce**
Steps to reproduce the behavior:
1. Click the Link https://github.com/isg-stuttgart/vscode-isg-cnc/blob/main/docs%5Cimages%5CsubProgramComment.png

**Expected behavior**
Show the picture instead of a 404.

**Screenshots**
[image]

**Desktop (please complete the following information):**
 - OS: Archlinux
 - Browser: Firefox
 - Version: 132.0.2

**Additional context**
Why is any information about my phone needed?

jurekseverin-isg, 2024-11-26:
Closed with https://github.com/isg-stuttgart/vscode-isg-cnc/pull/166

jurekseverin-isg, 2024-11-26:
@andreassteding-isg Thanks for reporting. The Problem is fixed and the picture will be visible on the next release.

closed 2024-11-26 by jurekseverin-isg